badgeteam/why2025-hardware #25: Unstable power rail / USB device port connection

renzenicolai, 2024-06-19:
Needs some more testing and research, the VSYS rail seems very unstable.

renzenicolai, 2024-07-14:
Turns out you can configure the current taken from the usb device port on the PMIC. We did that and now it's stable.

ranzbak, 2024-07-22:
Close this one in that case?

renzenicolai, 2024-07-22:
Nah, needs some more playing around with. For example the USB-A port can still cause the P4 do go into brownout so there is more stuff wrong here. I think it's mainly to do with the voltage on the VSYS rail still. Maybe we should look for a DC-DC for the 3.3v rail that can handle lower input voltages.

renzenicolai, 2024-07-27:
Check that the DC-DC has a low enough minimum input voltage to be able to use the whole battery.

closed 2024-08-21 by renzenicolai