badgeteam/why2025-hardware #33: Check protection circuit

renzenicolai, 2024-07-27:


ranzbak, 2024-07-27:
The rail to in v-in is ~1.5V for the LM339, when supplied with 3.3V, the max is 1.8V that can be compared.
So a rail to rail sulution is needed.

LCSC nummer: C4944986
Not too expensive: https://www.lcsc.com/product-detail/Operational-Amplifier_UMW-Youtai-Semiconductor-Co-Ltd-LMV324IDR_C4944986.html
Datasheet: https://www.lcsc.com/datasheet/lcsc_datasheet_2312211124_UMW-Youtai-Semiconductor-Co---Ltd--LMV324IDR_C4944986.pdf

ranzbak, 2024-07-29:
The 1V reference was pulled down to 1V.
Looked at voltage references to provide the 1V, they are pretty expensive.
When looking into the problem further we are trying to solve, protecting the chips from over voltage limiting the voltage is enough.
Given that the p4 at max can deliver 100mA current, a Zener that clamps the voltage is sufficient to protect the IC's involved.

Long story short, replaced the VPSRAM, VMIPI, VFLASH rail protection with zener diodes.
VESPHP is unprotected as the voltage is set by a resistor devider, en de EN_DCDC P4 can already disable the U3 vcore regulator if it's unhappy.

ranzbak, 2024-07-29:
https://github.com/badgeteam/why2025-hardware/pull/39

closed 2024-08-21 by renzenicolai