badgeteam/why2025-hardware #59: VESPHP now is very close to 1.20V which is the max voltage, would 1,12 V be better?

ranzbak, 2024-09-06:
Replacing R40 with a 430K resistor would put the set voltage at 1,12V.

In the ESP32 P4 sheet (V0.4), page 62 bottom of the page for reference.

arturo182, 2024-09-08:
Surprisingly, the official Espressif EV Board also uses 2x499K:

[image]

From https://docs.espressif.com/projects/esp-dev-kits/en/latest/_static/esp32-p4-function-ev-board/schematics/esp32-p4-function-ev-board-schematics.pdf

This of course is:

$$0.6V * (1 + \frac{499K}{499K}) = 1.2V$$

My guess would be this is because the ESP32-P4 uses the `FB_DCDC` pin to adjust the voltage as needed so the default of 1.2V as a starting point is not a problem for them?

renzenicolai, 2024-09-11:
Yeah, I checked and the rev 0.1 version of the development board still has it like this in the schematic, let's copy what Espressif does 😄

closed 2024-09-11 by renzenicolai