badgeteam/why2025-hardware #69: Add pull-up resistor to GPIO34

renzenicolai, 2024-09-30:
For using the JTAG interface GPIO34 (E2 on the internal expansion port) can be used to select between USB jtag and the JTAG pins on the external expansion port.

Add a pull-up resistor to E2 (GPIO34) to select USB jtag by default on boards where this fuse has been burned.

 - burn JTAG_SEL_ENABLE to 1
 - pull E2 (GPIO34) low
 - reset ESP32-P4

Maybe we could add a footprint for a 10-pin JTAG header too somewhere?

renzenicolai, 2024-09-30:
Extra ideas:
 - Add testpoint to P4 reset pin next to the RX and TX pins
 - Add solder jumper for shorting E2 to ground

ranzbak, 2024-10-04:
Pullup resistor 51K added to GPIO34.

Leaving the ticket open for a bit to see if there is space for a 10 pin JTAG header.

renzenicolai, 2024-10-06:
Continued in #82

closed 2024-10-06 by renzenicolai