badgeteam/why2025-hardware #77: Add PMID SW diode for better stability and efficiency of the PMIC

ranzbak, 2024-10-05:
SW to PMID diode Help to improve the efficiency. Recommend for applications with greater than 2.4A boost mode ouput.

From this document : https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/196/BQ25895_5F00_SchematicChecklist-V1p1.pdf

Since we see VBatt / VSYS interactions, and have problems at high current, we might want to add this diode.

[image]

ranzbak, 2024-10-05:
In commit d410a64c0f7fcd8e44ddb87f60cdcf2cb8bcdb74 the Diode has been added, this should at least make the OTG power more reliable.

renzenicolai, 2024-10-06:
Seems fixed.

closed 2024-10-06 by renzenicolai